xmos/sw_usb_audio #159: S/PDIF transmit discontinuities

danielpieczko, 2023-05-23:
Using the new S/PDIF transmit code from lib_spdif, output tests sometimes report discontinuities:
```
Channel 8: discontinuity (samples -5121286, -5121272 do not differ by 7 but by 14)
Channel 9: discontinuity (samples -3532169, -3532184 do not differ by -5 but by -15)
```
In this example, one sample went missing on channel 8, and two samples on channel 9. In the reproductions I've seen, it is usually one or two samples on each channel that are missing. Sometimes there is an invalid sample:
`Channel 9: discontinuity (samples -1243255, -1243268 do not differ by -5 but by -13)`
(expected samples: -1243265, -1243270)

So far, this has only been seen at sample rates of 176400 and 192000.

Running output tests with a three minute duration, this error is seen 10% of the time. When I've analysed how far into the streams the discontinuity occurs, I'll update here.

On the 316 board, it was most commonly seen on the `2AMi8o10xxsxxx` config, but on others as well.

On the 216 board, it was seen only on the `2AMi8o10xxsxxx_mix8` config.

xross, 2023-05-23:
Since this feels like a timing issue I've pushed some basic optimisations to the S/PDIF transmitted on the head of develop

xross, 2023-05-23:
Looking at the new implementation, I'm wondering if this `static int lastbit` should be reset to 0 on SR change
https://github.com/xmos/lib_spdif/blob/30f0039ae9d6adad16700b612ed2cea32995b40d/lib_spdif/src/SpdifTransmit.xc#L102

xross, 2023-05-23:
> Looking at the new implementation, I'm wondering if this `static int lastbit` should be reset to 0 on SR change @XMOS-JoeG
> 
> https://github.com/xmos/lib_spdif/blob/30f0039ae9d6adad16700b612ed2cea32995b40d/lib_spdif/src/SpdifTransmit.xc#L102

This would only be important if we wanted to make the start of stream predicable, for testing etc.

danielpieczko, 2023-05-30:
> Since this feels like a timing issue I've pushed some basic optimisations to the S/PDIF transmitted on the head of develop

Following these changes in lib_spdif, we no longer see the discontinuities reported in this issue where one or two samples are missing (across hundreds of ouput test attempts).

closed 2023-05-30 by danielpieczko