xmos/sw_usb_audio #162: S/PDIF output test sees unexpected zero sample values

danielpieczko, 2023-05-30:
On the xk_316_mc board, with lib_spdif:30f0039 the S/PDIF output test reports discontinuites where the sample value is zero. We don't know if this is an issue with the sender in sw_usb_audio or the receiver in sw_audio_analyzer.

Added some extra debug to sw_audio_analyzer to report samples after a discontinuity
```
E           Channel 8: step = 7
E           Channel 8: discontinuity 136087, 0 diff -136087 not 7 (sample 799607)
E           Channel 8: 0 0 0 0 0 0 0 0
E           Channel 8: 0 0 0 0 0 0 0 0
E           Channel 8: 0 0 0 136094 136101 136108 136115 136122
E           Channel 8: 136129 136136 136143 136150 136185 136192 136199 136206
E           Channel 8: 136213 136220 136227 136234 136297 136304 136311 136318
E           Channel 8: 136325 136332 136353 136360 136367 136374 136381 136388
E           Channel 8: 136395 136402 136409 136451 136458 136465 136472 136479
E           Channel 8: 136486 136493 136500 136521 136556 136563 136570 136577
E           Channel 8: 136584 136591 136598 136605 136647 136654 136661 136668
E           Channel 8: 136675 136682 136689 136696 136752 136759 136766 136773
E           Channel 8: 136780 136787 136794 136815 136836 136843 136850 136857
E           Channel 8: 136864 136871 136892 136899 136906 136913 136920 136927
E           Channel 8: step = 7
```

- Sample 136087 was received as expected. The next expected sample is 136094, but before this was received there were 19 samples of zero.
- There are some missing samples, eg. 136150 is followed by 136185 (four samples missing)
- After this discontinuity, the signal then runs for about three minutes without any errors
- Currently collecting more data on how often this happens, etc. but this example was at 176400 (can't tell how far into the output this occured, but will update later when that can be found)
- There could be parity errors and dropped samples - these are not reported by the test harness

danielpieczko, 2023-05-31:
This appears to be very rare: so far reproduced about 1 in 100 attempts.

I now have two more examples of this failure, both at 192kHz. In one case, there were at least 96 zero samples (that's the point where I stopped printing them out), and in the other there were 23/25 (different on the two channels).

danielpieczko, 2023-06-05:
Running four analogue output channels along with the S/PDIF channel pair, the analogue signal doesn't glitch at the same time as the S/PDIF discontinuity.

xross, 2023-06-16:
We plan to run repeat test with the old (<v5) S/PDIF transmitter in order to determine if the issue lies there.

danielpieczko, 2023-06-16:
It has been reproduced with the old S/PDIF transmit code.