xmos/sw_usb_audio #170: README.rst specifies broken link to "lib_xua"

harvv, 2023-06-13:
In the readme, section "Required Software (dependencies)" specifies 
  "lib_xua ([www.github.com/xmos/lib_xua](http://www.github.com/xmos/lib_xua))" .

This link leads to a Not Found Github page. Searching xmos repositories also yields no such "lib_xua" repository

Furthermore, references into subdirectories of this seemingly-missing lib_xua component are in multiple places in various files of the sw_usb_audio repo.

This is considerably confusing for first-timers trying to understand and build the code.

xross, 2023-06-13:
Thanks for the report. The repo will be made public in the near future. In the meantime please refer to the the lib_xua directory in the sw_usb_audio release zip.

xross, 2023-06-15:
[www.github.com/xmos/lib_xua](url) is now public