xmos/sw_usb_audio #205: Unable to change sample rate in Windows Sound menu

danielpieczko, 2024-01-10:
When using the Microsoft `usbaudio2` driver instead of the Thesycon driver, there are problems with the configuration of the device in the Windows sound menu:
- for all configs with more than two channels, the playback sample rate cannot be changed: the drop-down menu is greyed-out
- for `2AMi2o2xxxxxx`, the playback sample rate can be changed, but the menu is limited to `16-bit 44100`, `24 bit 44100` and `16-bit 48000`
- for all configs, the recording sample rate list has all frequencies, but only 24-bit, not 16-bit

With the Thesycon driver installed (but running exactly the same firmware on the MCAB), both the playback and record menus contain all sample rates and all bit depths.

danielpieczko, 2024-01-10:
USB trace of enumeration for `2AMi2o2xxxxxx` config
[2AMi2o2xxxxxx.zip](https://github.com/xmos/sw_usb_audio/files/13887133/2AMi2o2xxxxxx.zip)

The "get range" for sample rates gets the full list, and it sequentially sets the current sample rate to each of the values in the range. Yet the drop-down menu doesn't have all the options.

xross, 2024-01-10:
> * for all configs, the recording sample rate list has all frequencies, but only 24-bit, not 16-bit

16 bit interfaces are only exposed for playback, not recording

xross, 2024-01-10:
> * for `2AMi2o2xxxxxx`, the playback sample rate can be changed, but the menu is limited to `16-bit 44100`, `24 bit 44100` and `16-bit 48000`

Could not reproduce on Lenovo ThinkBook 13s-IML with Win10 Pro version 10.0.19045 build 1945

danielpieczko, 2024-01-10:
>     * for `2AMi2o2xxxxxx`, the playback sample rate can be changed, but the menu is limited to `16-bit 44100`, `24 bit 44100` and `16-bit 48000`

Also reproduced on my Windows 10 HP laptop, OS build 19045.3930

VergilWang15, 2024-01-18:
> When using the Microsoft `usbaudio2` driver instead(相反) of the Thesycon driver, there are problems with the configuration of the device in the Windows sound menu:
> 
> * for all configs with more than two channels, the playback sample rate cannot be changed: the drop-down menu is greyed-out
> * for `2AMi2o2xxxxxx`, the playback sample rate can be changed, but the menu is limited to `16-bit 44100`, `24 bit 44100` and `16-bit 48000`
> * for all configs, the recording sample rate list has all frequencies, but only 24-bit, not 16-bit
> 
> With the Thesycon driver installed (but running exactly the same firmware on the MCAB), both the playback and record menus contain all sample rates and all bit depths.

I encontered the exactly same problems on my Windows 11 Lenovo laptop with OS build 22621.3007.