xmos/sw_usb_audio #232: app_usb_aud_xk_evk_xu316: ET_ILLEGAL_RESOURCE is triggered on DoP flag detection

michele-perrone, 2024-05-16:
## Problem description
I'm trying to enable stereo DSD DoP in `sw_usb_audio` for the `XK-EVK-XU316 board`, according the latest `lib_xua` manual:
- The `XTC` version is 15.2.1
- The `sw_usb_audio` version is 8.0.0

After recompiling with DoP support, any attempt to play it crashes the XMOS. At the same time, PCM works without issues.

## Steps to reproduce
1. Download and unzip sw_usb_audio release 8.0.0: https://github.com/xmos/sw_usb_audio/releases/download/v8.0.0/sw_usb_audio_8.0.0.zip
2. Change current working directory to  `sw_usb_audio_8.0.0/sw_usb_audio/app_usb_aud_xk_evk_xu316`
3. Add the following lines in `src/core/xua_defs.h` to enable DoP:
```
#define NATIVE_DSD         (0)  /* Support only DoP */
#define DSD_CHANS_DAC      (2)  /* Stereo output */
#define PORT_DSD_DAC0      PORT_I2S_DAC_DATA
#define PORT_DSD_DAC1      PORT_I2S_ADC_DATA
#define PORT_DSD_CLK       PORT_MCLK_IN
```
4. Compile with `xmake all`
5. Run with `xscope`:
```
xrun --xscope bin/2AMi2o2xxxxxx/app_usb_aud_xk_evk_xu316_2AMi2o2xxxxxx.xe
```
6. Play a DSD64 or DSD128 file with DoP (tested with Audirvana Studio and foobar2000 on Windows 11)

## Expected result
I would expect `PORT_DSD_DAC0` to start outputting DoP data.

## Actual result
The DoP flags are recognized, after which the following error is thrown and the XMOS crashes. 
```
xrun: Program received signal ET_ILLEGAL_RESOURCE, Resource exception.
      [Switching to tile[1] core[0] (dual issue)]
      ConfigAudioPorts (p_i2s_dac=@0x82af8, numPortsDac=2, p_i2s_adc=@0x82b70, numPortsAdc=1, divide=8, clk_audio_bclk=774, p_mclk_in=66304, p_bclk=66304, p_lrclk=0, curSamFreq=<value optimized out>) at /home/michele/Projects/DWE-GIT/G26/sw_usb_audio_8.0.0/lib_
      xua/lib_xua/src/core/ports/audioports.xc:39
      39                asm volatile ("setc res[%0], %1" :: "r" (p_lrclk), "r" (0x200006));
```

michele-perrone, 2024-05-23:
See the issue in `lib_xua`: https://github.com/xmos/lib_xua/issues/394

xross, 2024-05-23:
Fixed in https://github.com/xmos/lib_xua/pull/395

closed 2024-05-23 by xross