xmos/sw_usb_audio #257: Compilation error  after #define  HID_CONTROLS (1)

jian-yuan-cmd, 2024-07-26:
the  code   can be compile complete in the directory  sw_usb_audio-_sw_v8_1_0/sw_usb_audio/app_usb_aud_xk_316_mc

but it can't be compile complete after I enable HID_CONTROLS. ( #define HID_CONTROLS (1) in  xua_conf.h )

the information of  compile error  is  below:

please help me check it .

thanks.

yuanjian@YuanJians-MacBook-Air app_usb_aud_xk_316_mc % xmake CONFIG=2AMi2o2xxxxxx -j
Checking build modules
Using build modules: lib_sw_pll(2.2.0) lib_xua(4.1.0) lib_i2c(6.2.0) lib_adat(1.2.0) lib_locks(2.2.0) lib_logging(3.2.0) lib_mic_array(4.6.0) lib_spdif(6.1.0) lib_xassert(4.2.0) lib_xud(2.3.1) lib_dsp(6.3.0)
Creating dependencies for pdm_rx.S
Creating dependencies for decimate_to_pcm_4ch.S
Creating dependencies for fastmix.S
Creating dependencies for mic_array_dual.xc
Creating dependencies for hires_delay.xc
Creating dependencies for decimator_interface.xc
Creating dependencies for usb_midi_support.xc
Creating dependencies for usb_midi.xc
Creating dependencies for hid.xc
Creating dependencies for dfu.xc
Creating dependencies for audioports.xc
Creating dependencies for pdm_mic.xc
Creating dependencies for mixer.xc
Creating dependencies for xua_ep0_uacreqs.xc
Creating dependencies for sw_pll_wrapper.xc
/Users/yuanjian/workspace/xmos/111_open/SDK/sw_usb_audio-_sw_v8_1_0/lib_xua/lib_xua/src/core/mixer/mixer.xc:284: warning: #warning OUT Vols in mixer, AFTER mix & map

    #warning OUT Vols in mixer, AFTER mix & map
Creating dependencies for clockgen.xc
/Users/yuanjian/workspace/xmos/111_open/SDK/sw_usb_audio-_sw_v8_1_0/lib_xua/lib_xua/src/core/mixer/mixer.xc:284: warning: #warning OUT Vols in mixer, AFTER mix & map

    #warning OUT Vols in mixer, AFTER mix & map
Creating dependencies for ep_buffer.xc
Creating dependencies for decouple.xc
Creating dependencies for xua_audiohub.xc
Creating dependencies for audiohub_initport.xc
Creating dependencies for warnings.xc
Creating dependencies for main.xc
Creating dependencies for audiohw.xc
Creating dependencies for XUD_HAL.xc
Creating dependencies for hid_report.c
Creating dependencies for flashlib_user.c
Creating dependencies for flash_interface.c
/Users/yuanjian/workspace/xmos/111_open/SDK/sw_usb_audio-_sw_v8_1_0/lib_xua/lib_xua/src/hid/hid_report.c:14:10: fatal error: 'hid_report_descriptor.h' file not found
#include "hid_report_descriptor.h"
         ^
1 error generated.
Creating dependencies for xuduser.c
Creating dependencies for audioports.c
Creating dependencies for pdm_mic_user.c
Creating dependencies for xua_endpoint0.c
Creating dependencies for vendorrequests.c
Creating dependencies for get_adc_counts.c
Creating dependencies for decouple_interrupt.c
Creating dependencies for hid_report.c
/Users/yuanjian/workspace/xmos/111_open/SDK/sw_usb_audio-_sw_v8_1_0/lib_xua/lib_xua/src/hid/hid_report.c:14:10: fatal error: 'hid_report_descriptor.h' file not found
#include "hid_report_descriptor.h"
         ^
1 error generated.
Compiling decouple_interrupt.c
Compiling get_adc_counts.c
Compiling vendorrequests.c
Compiling xua_endpoint0.c
Compiling pdm_mic_user.c
Compiling audioports.c
Compiling xuduser.c
Compiling flash_interface.c
Compiling flashlib_user.c
Compiling hid_report.c
Compiling XUD_HAL.xc
Compiling audiohw.xc
/Users/yuanjian/workspace/xmos/111_open/SDK/sw_usb_audio-_sw_v8_1_0/lib_xua/lib_xua/src/hid/hid_report.c:14:10: fatal error: 'hid_report_descriptor.h' file not found
#include "hid_report_descriptor.h"
         ^
1 error generated.
xmake[1]: *** [.build_2AMi2o2xxxxxx/_l_xua/src/hid/hid_report.c.o] Error 1
xmake[1]: *** Waiting for unfinished jobs....
Compiling main.xc
xmake: *** [bin/2AMi2o2xxxxxx/app_usb_aud_xk_316_mc_2AMi2o2xxxxxx.xe] Error 2
yuanjian@YuanJians-MacBook-Air app_usb_aud_xk_316_mc %

jian-yuan-cmd, 2024-08-13:
I have copied HID  code from other projects

jian-yuan-cmd, 2024-10-16:
I need your help to solve this HID issue.
(I copied the HID code from elsewhere, but after setting NUM-USB_CHAN-IN to 0,  it can't  report HID messages.)
Thank you.

jian-yuan-cmd, 2024-10-17:
reopen this issue, since this problem can be encountered by user directly, need to solve in the next release, thanks

jian-yuan-cmd, 2024-10-24:
I found that:
the  enum USBEndpoint Number_In （in file lib_xua/lib_xua/src/core/endpoint0/describor_defs. h
﻿
doesn't match the array XUD_EpType epTypeTableIn [ENDPOINT COUNT-IN] (in file lib_xua/lib_xua/src/core/main.xc)
﻿

After modification, even if NUM-USB_CAN-IN is set to 0, the HID Msg can still be reported to Host

shuchitak, 2025-03-14:
Hi @jian-yuan-cmd,
I tried recreating this by adding a NUM_USB_CHAN_IN=0 and HID_CONTROLS=1 config to app_usb_aud_xk_216_mc in sw_usb_audio. 
I had to make a small change (an include file was missing) to get it compiling but after it compiles, it runs fine for me. Button presses are generating non zero HID IN reports.

This is all I changed in sw_usb_audio.

```
diff --git a/app_usb_aud_xk_216_mc/configs_partial.cmake b/app_usb_aud_xk_216_mc/configs_partial.cmake
index cdcfdd5..d6f0d7c 100644
--- a/app_usb_aud_xk_216_mc/configs_partial.cmake
+++ b/app_usb_aud_xk_216_mc/configs_partial.cmake
@@ -14,6 +14,8 @@ set(APP_COMPILER_FLAGS_1AMi2o2xxxxxx ${SW_USB_AUDIO_FLAGS} -DAUDIO_CLASS=1)
 
 set(APP_COMPILER_FLAGS_2AMi8o8xxxxxx ${SW_USB_AUDIO_FLAGS})
 
+set(APP_COMPILER_FLAGS_2AMi0o8xxxxxx_hid ${SW_USB_AUDIO_FLAGS} -DHID_CONTROLS=1 -DNUM_USB_CHAN_IN=0)
+
 set(APP_COMPILER_FLAGS_2ASi8o8xxxxxx_tdm8 ${SW_USB_AUDIO_FLAGS} -DXUA_PCM_FORMAT=XUA_PCM_FORMAT_TDM
                                                                 -DMAX_FREQ=96000 -DCODEC_MASTER=1
                                                                 -DOUT_VOLUME_IN_MIXER=0)
diff --git a/app_usb_aud_xk_216_mc/src/extensions/hidbuttons.xc b/app_usb_aud_xk_216_mc/src/extensions/hidbuttons.xc
index 1f199ae..7703dee 100644
--- a/app_usb_aud_xk_216_mc/src/extensions/hidbuttons.xc
+++ b/app_usb_aud_xk_216_mc/src/extensions/hidbuttons.xc
@@ -7,6 +7,7 @@
 #include "app_usb_aud_xk_216_mc.h"
 #include "user_hid.h"
 #include "xua_hid_report.h"
+#include "xua_conf_full.h"
 
 #if HID_CONTROLS > 0
```

If this is still an issue for you, can you provide the exact config that you're testing? Thanks.

shuchitak, 2025-04-15:
Hi @jian-yuan-cmd, PR https://github.com/xmos/lib_xua/pull/455 has some HID related fixes.
- Get NUM_USB_CHAN_IN=0, NUM_USB_CHAN_OUT=0 with HID_CONTROLS=1 config compiling
- Fix HID with AUDIO_CLASS=1
- Fix for failing USB3CV HID Descriptor Test

Also, https://github.com/xmos/sw_usb_audio/pull/299 adds HID support to app_usb_aud_xk_316_mc.

Both the PRs are merged to develop. Hopefully, this fixes the HID related issues you were seeing. Feel free to open a new issue if you still see any HID related failures.

closed 2025-04-15 by shuchitak