xmos/sw_usb_audio #260: MIDI Stress test failures

danielpieczko, 2024-08-07:
MIDI stress test failures are reproducible on both Mac hardware agents, not every time but usually failing within a few attempts. There are two failures which don't look exactly the same, so there could be multiple issues here.

Either see glitches:
```
Failed: Failure at sample rate 192000
ERROR: Channel 0: Glitch detected, average non peak -76.820651, threshold -94.411780 (sample 294912)
ERROR: Channel 5: Glitch detected, average non peak -76.557674, threshold -94.350837 (sample 294912)
ERROR: Channel 1: Glitch detected, average non peak -78.162108, threshold -94.462208 (sample 294912)
ERROR: Channel 3: Glitch detected, average non peak -78.387714, threshold -94.472290 (sample 294912)
ERROR: Channel 4: Glitch detected, average non peak -76.216177, threshold -94.461089 (sample 294912)
ERROR: Channel 6: Glitch detected, average non peak -76.096009, threshold -94.404721 (sample 294912)
ERROR: Channel 7: Glitch detected, average non peak -76.128902, threshold -94.463358 (sample 294912)
ERROR: Channel 2: Glitch detected, average non peak -76.196843, threshold -94.405877 (sample 294912)
```

or incorrect frequencies and glitches:
```
Failed: Failure at sample rate 192000
ERROR: Channel 0 not expected freq (expected 1000 got 10) (sample 98304)
ERROR: Channel 1 not expected freq (expected 2000 got 10) (sample 98304)
ERROR: Channel 3 not expected freq (expected 4000 got 10) (sample 98304)
ERROR: Channel 2 not expected freq (expected 3000 got 10) (sample 98304)
ERROR: Channel 5 not expected freq (expected 6000 got 10) (sample 98304)
ERROR: Channel 6 not expected freq (expected 7000 got 10) (sample 98304)
ERROR: Channel 4 not expected freq (expected 5000 got 10) (sample 98304)
ERROR: Channel 7 not expected freq (expected 8000 got 10) (sample 98304)
ERROR: Channel 1: Glitch detected, average non peak -99.521303, threshold -102.325366 (sample 114688)
ERROR: Channel 4: Glitch detected, average non peak -99.606837, threshold -102.409805 (sample 114688)
ERROR: Channel 7: Glitch detected, average non peak -99.536749, threshold -102.434622 (sample 114688)
ERROR: Channel 3: Glitch detected, average non peak -99.053656, threshold -101.957031 (sample 114688)
ERROR: Channel 5: Glitch detected, average non peak -99.545772, threshold -102.661304 (sample 114688)
ERROR: Channel 2: Glitch detected, average non peak -99.189890, threshold -102.327940 (sample 114688)
ERROR: Channel 6: Glitch detected, average non peak -98.972413, threshold -102.364428 (sample 114688)
ERROR: Channel 0: Glitch detected, average non peak -99.452007, threshold -102.357930 (sample 114688)
Incorrect frequency on channel 0; got 10, expected 1000
Incorrect frequency on channel 1; got 10, expected 2000
Incorrect frequency on channel 2; got 10, expected 3000
Incorrect frequency on channel 3; got 10, expected 4000
Incorrect frequency on channel 4; got 10, expected 5000
Incorrect frequency on channel 5; got 10, expected 6000
Incorrect frequency on channel 6; got 10, expected 7000
Incorrect frequency on channel 7; got 10, expected 8000
```

A couple of examples of captured signals (expect sine waves):
[image]
[image]

danielpieczko, 2024-08-09:
This is seen with the xk_316_mc and xk_216_mc

danielpieczko, 2024-08-09:
I have reproduced this on xk_316_mc with the v8.1.0 release.
(Can't attempt a reproduction on xk_216_mc with that release because lib_xud v2.3.2 rise delay change is required to run the test)

xross, 2024-08-27:
This was found to be a test issue: https://github.com/xmos/sw_usb_audio/pull/266

shuchitak, 2025-09-10:
closing as fixed.

closed 2025-09-10 by shuchitak