xmos/sw_usb_audio #285: Consistency issue of USB endpoint enumeration numbering in SDK:sw_usb_audio-_sw_v8_1_0

jian-yuan-cmd, 2024-10-25:
In SDK: sw-usc_audio-sw_v8_1-0, 
array XUD_EpType epTypeTableIn [ENDPOINT COUNT-IN] doesn't match enum USBEndpointNumber_In, 
The current issue is that HID messages cannot be reported to host  when NUM-USB_CAN-IN is set to 0.
﻿Please ensure the consistency of USB endpoint numbers when releasing the new SDK. Thanks.

file: lib_xua/lib_xua/api/xua_conf_default.h
enum USBEndpointNumber_In
{
    ENDPOINT_NUMBER_IN_CONTROL,     /* Endpoint 0 */
#if (NUM_USB_CHAN_IN == 0) || defined (UAC_FORCE_FEEDBACK_EP)
    ENDPOINT_NUMBER_IN_FEEDBACK,   //---------------------------
#endif
#if (NUM_USB_CHAN_IN != 0)
    ENDPOINT_NUMBER_IN_AUDIO,     //-----------------------------
#endif
#if (XUA_SPDIF_RX_EN) || (XUA_ADAT_RX_EN)
    ENDPOINT_NUMBER_IN_INTERRUPT,   /* Audio interrupt/status EP */
#endif
#ifdef MIDI
    ENDPOINT_NUMBER_IN_MIDI,
#endif
#if XUA_OR_STATIC_HID_ENABLED
    ENDPOINT_NUMBER_IN_HID,
#endif
#ifdef IAP
#ifdef IAP_INT_EP
    ENDPOINT_NUMBER_IN_IAP_INT,
#endif
    ENDPOINT_NUMBER_IN_IAP,
#ifdef IAP_EA_NATIVE_TRANS
    ENDPOINT_NUMBER_IN_IAP_EA_NATIVE_TRANS,
#endif
#endif
    XUA_ENDPOINT_COUNT_IN           /* End marker */
};

file:lib_xua/lib_xua/src/core/main.xc
XUD_EpType epTypeTableIn[ENDPOINT_COUNT_IN] = { XUD_EPTYPE_CTL | XUD_STATUS_ENABLE,
                                            XUD_EPTYPE_ISO,  //--------------------------------

#if (NUM_USB_CHAN_IN == 0) || defined(UAC_FORCE_FEEDBACK_EP)
                                            XUD_EPTYPE_ISO,    /* Async feedback endpoint */ //----------------
#endif
#if (XUA_SPDIF_RX_EN || XUA_ADAT_RX_EN)
                                            XUD_EPTYPE_INT,
#endif
#ifdef MIDI
                                            XUD_EPTYPE_BUL,
#endif
#if XUA_OR_STATIC_HID_ENABLED
                                            XUD_EPTYPE_INT,
#endif
#ifdef IAP
                                            XUD_EPTYPE_BUL | XUD_STATUS_ENABLE,
#ifdef IAP_INT_EP
                                            XUD_EPTYPE_BUL | XUD_STATUS_ENABLE,
#endif
#ifdef IAP_EA_NATIVE_TRANS
                                            XUD_EPTYPE_BUL | XUD_STATUS_ENABLE,
#endif
#endif
                                        };

jian-yuan-cmd, 2024-10-25:
The 5.0.0 version of lib_xua appears to have resolved this issue.
However, there is still an issue of inconsistent USB endpoint numering.
To avoid potential risks, it is recommended to modify it to be consistent.thanks.

xross, 2024-10-25:
> To avoid potential risks, it is recommended to modify it to be consistent.thanks.
can you explain in a bit more detail what you mean with this please? Maybe with an example.

jian-yuan-cmd, 2024-10-28:
in usb audio project  code,   when we use c_xud_in[],  its index value depends on the corresponding enumeration value: for example,  c_xud_in[ENDPOINT_NUMBER_IN_HID],  
so , I suggest the sequence of XUD_EpType epTypeTableIn[ ]       ( defined in  file "lib_xua/lib_xua/src/core/main.xc" )
should keep uniformity with enum USBEndpointNumber_In   (defined in lib_xua/lib_xua/api/xua_conf_default.h)

for example ,  the value of ENDPOINT_NUMBER_IN_FEEDBACK is 1,
but  the corresponding  index in XUD_EpType epTypeTableIn[]   is 2.
We should modify one of the definition sequence.

xross, 2024-10-28:
These indices need to match, if they don't it's a bug for sure.