xmos/sw_usb_audio #290: Driver confusing info on xmos.com

QuinnWang, 2024-11-04:
https://www.xmos.com/software/usb-audio/driver-support/, it said:
Products based on USB Audio <= v8.1.0 use Thesycon <= v5.68.0
Products based on USB Audio >= v9.0.0 use Thesycon >= v5.70.0

But there is no v9.0.0 yet, and the latest 3 version EVAL driver in the list https://www.xmos.com/file/usb-audio-class-20-evaluation-driver-windows are:
- v5.58.0
- v5.57.0
- v4.13.0
So no v5.68.0 and v5.70.0 above, I think this makes us a bit confused, then suggested to change accordingly

xross, 2024-11-04:
HI Quinn, the website has gotten ahead of the USB Audio releases. Expect to see a v9 which requires Thesycon 5.70+ very soon!

You should use 5.58.0 eval for now, or if a getting a driver from Thesycon directly request <= 5.68. Thesycon are aware of the changes coming in v9 and will provide the correct driver version regardless

xross, 2025-01-08:
HI @QuinnWang are you happy with this information now?

QuinnWang, 2025-01-09:
All good now, thanks

closed 2025-01-09 by QuinnWang