xmos/sw_usb_audio #298: app_usb_aud_xk_evk_xu316: noise on inputs 1/2

aleksandrgilfanov, 2025-02-15:
I have xcore.ai board, and there is some strange noise on inputs 1/2. Nothing is connected into "LINE IN" jack.

reproducible on latest commit: 9596ec024d42aa66f579470e892907e1dffab30e, or sw_usb_audio-_sw_v8_1_0, or whatever

[image]

Probably it is issue of https://github.com/xmos/lib_board_support or even hardware desing problem...

xross, 2025-02-17:
- What build config are you using?
- Have you tried with the latest version of lib_board support?

aleksandrgilfanov, 2025-02-17:
> What build config are you using?

default Audio Class 2 config
```
# Audio Class 2, Async, I2S Master, 2xInput, 2xOutput
XCC_FLAGS_2AMi2o2xxxxxx = $(BUILD_FLAGS)
INCLUDE_ONLY_IN_2AMi2o2xxxxxx =
```

> Have you tried with the latest version of lib_board support?

yes, same on c73abb39efb13a1d23f1c1b9c06582615e7afb1d

elektron314, 2025-02-26:
That's exactly what I have on my board based on your xcore.ai devkit
Nothing on inputs electrically, -36dB of noise.

[image]

XMOS-JoeG, 2025-05-13:
Just checked XK-EVK-XU316 with no input using sw_usb_audio_9.0.0 (app_usb_aud_xk_evk_xu316_2AMi2o2xxxxxx.xe)

All looks fine. Attached plot from REW. Make sure all the sample rates are set to same values to avoid the OS doing any resampling (windows sound panel etc). I was testing with everything at 48kHz (win sound panel playback, recording, REW sample rate)

[image]

xross, 2025-05-13:
Since this appears to be due to incorrect host configuration, closing the issue.
It's worth noting that XMOS thorough test before release.

closed 2025-05-13 by xross