xmos/sw_usb_audio #300: ADAT-RX

Kevin1985309, 2025-03-19:
Hello ,
I am currently developing a device that can both send and receive ADAT signals. However, during testing, I have encountered some issues where the ADAT signal reception results in empty data. It appears that there might be some bugs in the source code provided by XMOS. Below are the details of my testing results:

Hardware:
XK-AUDIO-316-MC-AB
Windows PC

Software:
USB Audio Software version 8.1.0 or 9.0.0, specifically using app_usb_aud_xk_316_mc_2AMi16o16xxxaax.xe

Testing Procedure:
I played a 16-channel 1kHz sine wave signal using Adobe Audition and connected the ADAT-TX and ADAT-RX of the XK-AUDIO-316-MC-AB with a cable. I selected the XMOS ADAT Clock on the device driver side and recorded 16-channel audio data using Audacity. After recording for a while, data loss occurs, and at this point, all ADAT-RX channel data is empty. This issue typically reproduces within five minutes.

I have tested this on three different Windows PCs, and the problem persists, indicating that it is unlikely to be caused by the computer hardware.

Have you encountered this issue in your tests? If so, do you have any suggestions on how to resolve it?

Thank you for your assistance.

Best regards,

[image]

Kevin1985309, 2025-03-21:
This doesn't seem to be just an ADAT problem. 
When I use the source code to modify a 2-in 2-out firmware, short the input and output, and then record, I find that the same problem occurs. I can reproduce it once after 15 minutes of recording.

Kevin1985309, 2025-03-25:
So, using a single device to test ADAT transmission and reception is unreasonable. (This is usually not done; it only occurs during the software testing phase.)

By "shorting input and output," do you mean the I2S data lines?

Yes, I modified an I2S firmware for 2-in-2-out testing. At this point, it no longer involves the ADAT interface. During long-term I2S 2-in-2-out testing (on the same device), I encountered a situation where there was no data on Windows. This no longer involves the master-slave relationship because the device itself is the I2S master, and I2S in only needs to focus on data reception. (This might be a new issue.)
However, when I tested on a Mac computer, also using a single device and shorting the data lines, whether testing the 2-in-2-out I2S (with my modified 2-in-2-out firmware) or testing the ADAT source code (app_usb_aud_xk_316_mc_2AMi16o16xxxaax.xe), there was no occurrence of empty data.

It seems that the test results are inconsistent between Windows and Mac, which is quite puzzling to me.

In fact, I need a stable ADAT_RX. Initially, I tried testing with two boards, but the results were the same; they both exhibited data loss.

xross, 2025-03-25:
> So, using a single device to test ADAT transmission and reception is unreasonable. (This is usually not done; it only occurs during the software testing phase.)
> 

In general yes, clocking something from itself is likely to be unstable 

> In fact, I need a stable ADAT_RX. Initially, I tried testing with two boards, but the results were the same; they both exhibited data loss.

This would be unexpected, assuming you have the clock source set correctly.

Kevin1985309, 2025-03-26:
I conducted the following tests using two boards:

Test 1:
Board A is connected to Windows and is responsible for ADAT-TX, connected to Board B's ADAT-RX.
Board B is connected to a Mac and is responsible for ADAT-RX, with the ADAT clock source selected on the Mac.
Result of Test 1: The test ran for about 50 minutes, during which all statuses were normal, and no empty data issues were observed.

Test 2:
Board A is connected to Windows and is responsible for ADAT-TX, connected to Board B's ADAT-RX.
Board B is connected to Windows 11 and is responsible for ADAT-RX, with the ADAT clock source selected in the Windows driver.
Result of Test 2: An empty data issue was observed, which typically recurs within 20 minutes.

In the tests above, I repeatedly confirmed that the clock source selection for ADAT-RX was correct.

Could you help test this issue on Windows? Thank you. 
This might be a compatibility issue with Windows, and since there are a lot of Windows users, they might encounter the same problem.

Kevin1985309, 2025-03-26:
Yes, I am using ASIO, and I have set 2048 samples buffer on the ASIO driver.